sinara-hw/Zotino #2: v2.0 cyclic failure every ~15s

sbourdeauducq, 2018-05-21:
Every ~15s my Zotino stops producing an output signal for a few seconds, and the blank period is followed by a burst of noise:
[image]
This is with a kernel that produces a 1kHz tone:
```python
import struct

from artiq.experiment import *
from artiq.coredevice.ad53xx import *

class Tone(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("zotino0")

    @kernel
    def run(self):
        self.core.reset()
        self.zotino0.init()
        delay(1*ms)
        while True:
            self.zotino0.write_dac(1, 0.1)
            self.zotino0.load()
            delay(0.5*ms)
            self.zotino0.write_dac(1, -0.1)
            self.zotino0.load()
            delay(0.5*ms)
```

Even with nothing running on the core device, the bursts of noise are still observed, which points to a  hardware problem:
[image]

sbourdeauducq, 2018-05-21:
This behavior is observed on at least 2 channels. I don't have another board to test.

gkasprow, 2018-05-21:
Before I manage to recreate the issue, could you please have a look at the supply rails?
+/-12V which supplies DAC and buffers.

sbourdeauducq, 2018-05-21:
Where do I measure those? The PDF schematics aren't in https://github.com/sinara-hw/Zotino

sbourdeauducq, 2018-05-21:
Yep, the power supply is crapping out, here is the -12V:
[image]

gkasprow, 2018-05-21:
schematics are in [releases ](https://github.com/sinara-hw/Zotino/releases)
OK, so it seems to be the power supply issue. Please check if the power is problematic before the regulators.
What is the output load ?

sbourdeauducq, 2018-05-21:
> What is the output load ?

Open, I only have the scope (1M) connected.

gkasprow, 2018-05-21:
please measure V+ and V- with the scope
[image]

gkasprow, 2018-05-21:
please also check if the DAC od the LDO above N12V0A is getting hot.

sbourdeauducq, 2018-05-23:
The +12V is working correctly.
The -12V at the output of the LDO has the problem shown above, but its input is stable. It does get very hot. So I guess we are seeing the LDO thermal shutdown.

sbourdeauducq, 2018-05-23:
There is +/-15V at the inputs of the LDOs. Isn't that a bit high?
Also, the -12V glitches only begins to occur after the board has been running for a few minutes, which is consistent with thermal shutdown.

sbourdeauducq, 2018-05-23:
From a cursory inspection (no thermal camera here yet), nothing else on the board other than the DAC chip gets hot. I wonder if it may be drawing more than its specified current.

sbourdeauducq, 2018-05-23:
Contrary to what I thought, the DAC chip on my Zotino 1.1 is not burned, so I have added a heatsink to it and I am testing that board now. The switching regulator output is still +/-15V.

sbourdeauducq, 2018-05-23:
That 1.1 board doesn't have the problem. The -12V LDO is colder, too.

gkasprow, 2018-05-23:
we recreated same problem in the lab. the Zotino board was placed on the most right side of the crate where cooling is not sufficient.

sbourdeauducq, 2018-05-23:
The problem is systematically happening on the v1.2 board, when it is is exposed to air at ambient temperature without hot objects nearby. What should I do, turn on the TEC cooler?

sbourdeauducq, 2018-05-23:
Even with the board outside the crate, the bug is present.

gkasprow, 2018-05-23:
you can try to remove the Peltier module and attach the heatsink directly.

hartytp, 2018-05-23:
@gkasprow you think this is thermal runaway in the dac? That doesn't seem likely to me. On our boards, with no forced air, the dac is barely warm to touch. How hot is it on your board before this issue occurs?

gkasprow, 2018-05-23:
This happens when the DAC even for short moment gets its temperature exceeded. It gets damaged, it still works but consumes much more current so it enters this thermal trip quickly. 
Let's add thermal protection in next revision of Zotino which will switch off the power when temperature is i.e. 55 degrees and indicate it with LED.

hartytp, 2018-05-23:
@gkasprow did you test that by heating the dac with a hot air gun?

I just don't get this:
- iirc the dac is rated to normal operation at 85C die temp, and max tem 120C
- the datasheet has curves for current draw versus temperature which show no runaway
- the power consumption is low so the die should only be a few C hotter than the case (based on specifies thermal impedance)
- so unless the dac case / pcb is quite a bit hotter than 85C there should be no problem at all. On my board it just wasnt that hot. So why are we seeing this?
- I've used these DACs on eval boards in enclosures stabilised by resistive heating with no problems. Based on your explanation this should not be possible

So I think there must be some other cause for all of this. What do you think?

sbourdeauducq, 2018-05-23:
> Let's add thermal protection in next revision of Zotino which will switch off the power when temperature is i.e. 55 degrees and indicate it with LED.

If that's really the problem, this won't be a solution: the thermal protection will just be activated all of the time in the field. Again, the bug is present on my board *outside* the chassis, and it's not the hot HK weather, I had AC in the lab.

hartytp, 2018-05-23:
Note to self, let's also check that this isn't something dumb like we've used some DAC variant that's only specified over a very narrow temperature range.

gkasprow, 2018-05-23:
I also have AC in my lab and the failure happened today.
The SMPS delivers max +/- 15V so even if the LDO gets unstable, it is still within safe operation area.
I have the DAC devkit so will give it a try.

gkasprow, 2018-05-23:
I attached the heat sink directly to the DAC after removing the Peltier, the current consumption is stable and the DAC is cold. The board is on the bench and I have the air flow generated by 12V fan supplied from 10V.

gkasprow, 2018-05-23:
I removed the fan and the current consumption started growing. When it had grown by 20mA, I turned the fans on and it felt back to the original value.

jordens, 2018-05-23:
It might generally be wise to enable thermal shutdown in the AD5372 by default in ARTIQ.

hartytp, 2018-05-23:
> I removed the fan and the current consumption started growing. When it had grown by 20mA, I turned the fans on and it felt back to the original value.

Do you know from which rail the current was drawn?

hartytp, 2018-05-23:
Were your observations consistent with the data sheet?

[image]

hartytp, 2018-05-23:
> It might generally be wise to enable thermal shutdown in the AD5372 by default in ARTIQ.

We did consider that in the past (and it might not be a bad idea to do implement it), but the thermal shut-down only kicks in at 130C die temperature. IIRC, @gkasprow felt that by that point the DAC had been destroyed. Also, it's not great protection, as it doesn't help if the user powers the board on for a while before programming. Better to ensure that Zotino doesn't spontaneously die under normal use.

gkasprow, 2018-05-23:
I took the ADI devkit, powered it on and heated to 97 degrees. Did not observe any change in current consumption. -12V and +12V was below 10mA (bench PSU current display resolution)

hartytp, 2018-05-23:
Good!

So, either the DAC on Zotino has been damaged by something, which is leading to this behaviour. Or, we have some connection on Zotino that's different to the eval board. The question is which one? Maybe try disconnecting all the OpAmps from the DAC outputs?

gkasprow, 2018-05-23:
My DAC on Zotino is already broken. What I can do is to measure the output current on the RC filter resistors.
I will also check how the +/- 12V rails wake up.

jordens, 2018-05-23:
I am pretty sure that the thermal shutdown is designed to prevent and identify (since it doesn't clear itself) destructive temperature excursions. And I'd bet that if the damage is thermal, the shutdown would trip. But yes, I'd also bet that the problem is not bad cooling.

gkasprow, 2018-05-23:
the +/- 12V rails wake up in monotonic way. +12 wakes up faster than -12V.

hartytp, 2018-05-23:
> I am pretty sure that the thermal shutdown is designed to prevent and identify (since it doesn't clear itself) destructive temperature excursions. And I'd bet that if the damage is thermal, the shutdown would trip. But yes, I'd also bet that the problem is not bad cooling.

Exactly.

gkasprow, 2018-05-23:
I let the DAC to trip. Then I switched on the fan and the current consumption got lowered to initial value...
So it does not seem to be any latchup effect.

gkasprow, 2018-05-23:
There is one difference between eval kit and Zotino.
On Eval kit the pin 13 is grounded.
In datasheet is is marked as NC

gkasprow, 2018-05-23:
I connected it to GND but it did not help

gkasprow, 2018-05-23:
the DAC accepts up to 5V on digital supply. We have 3.3V supply which also supplies the LVDS drivers. And these drivers would be killed first.

gkasprow, 2018-05-23:
I'm just curious if this un-connected NC pin could kill the DAC. The reference can survive up to 40V on its input.
The DAC draws excessive current from its negative rail.

hartytp, 2018-05-23:
Sure, my point is just that characterizing the dead DAC carefully may not tell you anything about what killed it.

gkasprow, 2018-05-23:
I have one more Zotino that works correctly. I shorted pin 13 to GND and that's the only think I can do.
I can heat it to 90 degrees and see how it behaves.

gkasprow, 2018-05-23:
I posted it here
https://ez.analog.com/message/350275-failing-ad5372-dacs

dhslichter, 2018-05-23:
Looks like AD suggests powering up as -12V, +12V, +3.3V:

https://ez.analog.com/thread/47483

gkasprow, 2018-05-23:
I've noticed that. It is recommended pattern. We should fix it in next revision just in case.
I applied other sequences to the devkit and nothing happened.

hartytp, 2018-05-23:
@dhslichter I've seen that post before, but I don't think it applies to us. Importantly, Zotino should meet all the absolute maximum ratings by design. e.g. the digital sequencing shouldn't matter, since the inputs/outputs connect to ICs that run from the same regulator as Zotino.

Also, I've run eval boards for this DAC from bench top PSUs with no real care paid to power sequencing and never seen an issue like this.

hartytp, 2018-05-23:
> The reference can survive up to 40V on its input.

The data sheet says 5.5V max for VREF0 and VREF1

hartytp, 2018-05-23:
hmm I really can't think of anything to do here apart from removing all components from a working board until we strip it down to the very most basic design. If that works, then add things one by one and try to see which causes DAC death.

gkasprow, 2018-05-23:
Another observation - the board with issues, has them from very beginning. There are boards that work perfectly fine.
I tried with temperature and various supply sequences on devkit and they didn't harm it.
I will try the same with the board that works fine. But it's quite possible that we never find the problem in this way.
I'll do yet another thing - buy a few DACs and will work on the same board that causes issues.
I will replace the DAC on particular board and will see if the issue reappears. 
If not, I will play with voltage sequencing and logic levels during power up...

gkasprow, 2018-05-23:
@hartytp I mean 40V for reference IC.

dhslichter, 2018-05-23:
> Also, I've run eval boards for this DAC from bench top PSUs with no real care paid to power sequencing and never seen an issue like this.

Same here, I merely mentioned it in case it might make a difference.

sbourdeauducq, 2018-05-25:
Cooling the DAC chip works around the problem.

klickverbot, 2018-05-31:
For the record, one of our boards (v1.2) has developed this issue now as well (N12V0A dropping to ~2.5V every few seconds).

gkasprow, 2018-05-31:
I've ordered a bunch of DAC chips and will play with them to identify a moment where the DAC starts consuming excessive power on negative rail. Did not manage to do it with devkit.

cjbe, 2018-06-01:
To add to @klickverbot's comment, the v1.2 board we saw this problem on was new, and showed the problem within a few hours of powering it on for the first time (i.e. we powered it on, and saw the problem as soon as we looked for it).

There has been a v1.1 Zotino sitting in the same airflow within 10cm of the problematic v1.2 - this v1.1 Zotino has been working for a month with no issues.

jordens, 2018-06-01:
I enabled OT shutdown by default. This should at least keep the chips from dying in many cases. https://github.com/m-labs/artiq/commit/2c344686d9b0f8c3f8cc65db692468e7ac443310

sbourdeauducq, 2018-06-02:
Here is a picture of the DAC on the faulty board, in case this is an issue with a certain batch of chips.
[image]

hartytp, 2018-06-14:
@gkasprow any luck tracking this down. I'm going to buy a few of the DAC chips and have a look at the dead board here.

hartytp, 2018-06-14:
@gkasprow did you see that you got a follow up question from ADI? https://ez.analog.com/message/350275-failing-ad5372-dacs

hartytp, 2018-06-14:
@gkasprow Having said that, I don't see the BCPZ package stocked in any of the usual places on less than a 6 week lead time! Am I missing something?

[91 more replies]

closed 2018-08-14 by gkasprow