sinara-hw/Zotino #5: Switch DAC package

hartytp, 2018-06-18:
Looks like the AD5372BCPZ (LFCSP package) is being discontinued http://www.analog.com/media/en/PCN/ADI_PDN_16_0050_Rev_B_Form.pdf p47.

Replacement is the AD5372BSTZ (LQFP). Let's switch to that for the next revision as the availability seems to be better (although, the thermal impedance is marginally higher).

dhslichter, 2018-06-18:
Sounds good to me.

gkasprow, 2018-08-13:
well, they say that "PDN is no longer applicable for this part.It has been removed on this version of the PDN"
It's available now.

hartytp, 2018-08-14:
@gkasprow okay, if it's no longer depreciated then we can keep the current package. However, I still had the impression that the package we are using is harder to get hold of (fewer people seem to stock it), so we might want to change anyway. I'll leave that to you.

gkasprow, 2018-08-14:
I managed to place both footprints so any package can be installed.

closed 2018-08-14 by gkasprow

gkasprow, 2018-08-14:
[image]

hartytp, 2018-08-15:
:) that solves that problem