sinara-hw/Zotino #6: LDO reverse voltage diode

hartytp, 2018-06-27:
@gkasprow On one of my boards, I measure +0.7V on the N12V0 rail.

Looking again at the schematic, it seems that D11 and D12, which are supposed to prevent the LDOs from being reverse biased during turn on/off are Zener diodes. However, the DAC data sheet specifies an absolute maximum voltage on the N12V0 rail of +0.3V, so we're exceeding this and could easily destroy the DAC like that.

- [ ] D11 and D12 should be Schottky diodes. In the past, I've found it necessary to select diodes with very low turn on voltages to ensure that the diodes switch on before the protection diodes in the DAC (e.g. by +0.4V the DAC could be passing quite a lot of current through its internal diode junctions). Might also be worth adding a small resistor on the supply rails to help ensure that the external protection diode switches on hard before the internal ones do....

I have a feeling that this explains our mysterious DAC deaths, but I can't test that until the DAC goes back into stock somewhere...

hartytp, 2018-06-27:
NB all other Sinara boards should be checked for this issue.

klickverbot, 2018-06-27:
You mean D10 and D11 in v1.2rc1, right? Good catch – we should probably be using something like the [DB2S20500L](https://industrial.panasonic.com/content/data/SC/ds/ds4/DB2S20500L_E.pdf) (obviously part choice depends on how much current can be sourced).

hartytp, 2018-06-27:
@klickverbot right. Those look pretty good, but really anything chosen for low forwards voltage at a hundred mA or so. Ideally, we'd want something that can take the entire output of the LDO, so either parallel a few diodes, or find one with a low forward voltage and a higher max current.

hartytp, 2018-06-27:
@klickverbot do you have any of those in stock that I can nab?

gkasprow, 2018-06-27:
@hartytp This makes sense. In our case the part that may cause damage is AFE which consumes power between V+ and V- unlike the DAC itself. And the AFE current can bias the DAC in wrong way when one of rails wakes up a bit earlier.
That's why I did not manage to kill the devkit because there was no AFE that draws the current.
Any schottky diode with low Vf would work, even SS16 - they have similar characteristics as one posted above. The AFE-generated current can simply trigger latchup mechanism and negative supply current does damage.

The only fact that does not fit to this theory is that the prototype had independent regulators. But since the LDOs are mainly based on MOSFETs, they heve reverse diodes which managed to close the loop from the AFE rail to the DAC rail.

hartytp, 2018-07-03:
@klickverbot I ordered some SS14, which seem like a good choice for this.

gkasprow, 2018-07-10:
I installed 6V8 Zener diode in parallel with negative LDO on 4 boards + 2x SK26 on top of TVS diodes.
It works, no excessive current is drawn.

gkasprow, 2018-07-18:
Fully agree.

closed 2018-08-14 by gkasprow