sinara-hw/Zotino #7: Zotino rev1.1 issue

prasath-d, 2018-07-03:
Hello,
I am using zotino rev1.1 with kasli as a controller.
I tried to set the voltages in the DAC channel by using the code given in the demo program.
```
import numpy as np
from artiq.experiment import *

class zotino_demo(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("zotino0")

    @kernel
    def run(self):
        self.core.reset()
        self.zotino0.init()
        delay(100*ms)
        while True:
            self.zotino0.set_leds(0xff)
            delay(100*ms)
            self.zotino0.set_dac(
                [-9. + i*.5 for i in range(32)],
                list(range(32))
            )

```
The output I got from one of the channel was
[image]
There is a deadtime of 2s between every 600ms.
Then tried to understand the problem by modifying the code.
```
import numpy as np
from artiq.experiment import *

class zotino_demo2(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("zotino0")

    @kernel
    def run(self):
        self.core.reset()
        delay(100*ms)        
        self.zotino0.init()
        while True:
            self.zotino0.set_leds(0xff)
            delay(10*ms)
            self.zotino0.set_dac(
                [-9. + i*.5 for i in range(32)],
                list(range(32))
            )
            delay(10*ms)
            self.zotino0.set_dac(
                [9. - i*.5 for i in range(32)],
                list(range(32))
            )
            delay(10*ms)

```
This switches the output voltage every 10ms from positive to negative and, I saw the same
[image]
The output works for 600ms and stops working for 2s.

I recorded the output, when no program is running.
[image]
where, there is a noise output of around ~10mv which is also having 600ms width and 2s of deadtime.
could you please help me in understanding this issue. is it a hardware issue or I am missing something?
Thank you

hartytp, 2018-07-03:
https://github.com/sinara-hw/Zotino/issues/2 :(

closed 2018-07-03 by hartytp

jbqubit, 2018-07-10:
cf https://github.com/sinara-hw/Zotino/issues/2#issuecomment-403926712