sinara-hw/Zotino #10: SMPS voltage incorrect

hartytp, 2018-07-10:
@gkasprow the SMPS should be putting out +-13V. It seems to put out +-15V. The same happens on Sampler v2.0. Can you reproduce this on your board?

gkasprow, 2018-07-10:
I've noticed that recently. 
The datasheet specifies it as:
Ra = 53.55 / (Vout - 12.02) - 18
We want +/- 13V so Ra = 36R and such resistor is installed.
I will play with this resistor.
AFAIR in previous series I measured +/- 13V....

hartytp, 2018-07-10:
> I've noticed that recently.

Okay, glad you see this too so it's not just me!

> We want +/- 13V so Ra = 36R and such resistor is installed.
> I will play with this resistor.

Yes, AFAICT the datasheet is correct. So, it's not clear to me why we're getting the wrong voltage.

hartytp, 2018-07-11:
@gkasprow  ooops :)

the units of the 36 resistance are kOhms, not Ohms. Which, when one thinks about it is kind of obvious; the drive current on that pin is probably only uA, which is pretty hard to measure with a 36Ohm drive resistor.

hartytp, 2018-07-11:
Confirmed on two boards: Ra=36k gives +-13V.

This isn't really a problem as using +-15V only increases the power consumption a bit. So, while we should fix this for the next revision, I don't think it's worth patching existing boards. It does mean that the max current the SMPS can supply will be a bit lower than it would otherwise be...

closed 2018-08-14 by gkasprow