sinara-hw/Zotino #13: SMPS over current

hartytp, 2018-07-11:
@gkasprow the SMPS is only rated to withstand over-current conditions for up to 30s. After that it can be damaged. Are we sure that we're okay with that. e.g. this means the SMPS can easily be damaged if the SCSI connector draws too much current.

hartytp, 2018-07-11:
@gkasprow what do you want to do about this?

My worry is a situation like this:
- someone connects the outputs of Zotino to a capacitive load, so the DC load current is 0mA
- someone draws some current from the SCSI connector
- no they start changing the outputs rapidly, so the AC current draw is high. At this point they hit an over current condition

Or, simply the fact that the SCSI +-13V outputs are not properly current limited and shorting them for more than 30s could damage the board.

So, I think we should either use a bigger SMPS (and, maybe one with more robust protection), or do something like:
- do not connect the +-13V to the SCSI/IDCs by default. Instead, add DNF 0R resistors that must be populated to connect the power
- add a clear warning to the power budget/connector page explaining all of this

We should also add a note on the power page that the SMPS cannot withstand an indefinite over current condition.

gkasprow, 2018-07-11:
We can add PTC fuses rated for i.e. 50mA.

gkasprow, 2018-07-11:
Since external AFE is not yet defined I'm not sure if it's worth to provide symmetrical power supply from Zotino. There will always be a case that it is not enough. Over-rating SMPS on Zotino would be costly.
Maybe just leave +12V rail and install dedicated SMPS on remote AFE? Or get rid of any supply..

hartytp, 2018-07-11:
> Maybe just leave +12V rail and install dedicated SMPS on remote AFE? Or get rid of any supply..

I think that one of those two options would be better.

gkasprow, 2018-07-11:
So let's put 0R jumper so when needed, 12V can be used to supply to remote AFE.