sinara-hw/Zotino #18: v1.3 review

gkasprow, 2018-08-15:
I drafted new [release](https://github.com/sinara-hw/Zotino/releases/tag/Zotino%2Fv1.3)
@hartytp @jordens please have a look.

gkasprow, 2018-08-15:
I've just noticed that version on schematics is 1.2 instead of 1.3

gkasprow, 2018-08-15:
I also published short form schematics, but without assembly variants.
Altium gives two options - original schematics or physical schematics but with assembly variants.

hartytp, 2018-08-15:
Look good, thanks!

That looks really good, thanks again for all your hard work on this! FWIW, I'm considering v1.3 to be the first revision that's "production" rather than "prototype".

- on the schematic, the shield clips aren't labeled DNF. Should they be?
- IIRC, the reason you didn't 0-index the channels on the HD68 and IDCs was some Altium issue. That's totally fine (I'm just adding this note as a reminder to myself)
- were we planning to have TVS as well as Schottky diodes on this design? Or, not bother?

gkasprow, 2018-08-15:
@hartytp as I wrote, look at the right schematics. There are two versions.
TVS were only to protect against feeding power from AFE.
Since we won't supply AFE from Zotino, they are not needed.
0-indexing is Altium feature. We can migitate it by placing 32 copies of the output_channel block on the schematic.

gkasprow, 2018-08-15:
I do not change designators/properties to DNF because during library update they are reverted back to original values. Instead I use either no_value components or assembly variants.

hartytp, 2018-08-15:
thanks for the clarifications Greg.

That all looks good to me!

Closing this issue, but others should feel free to reopen if they spot anything.

closed 2018-08-15 by hartytp

jordens, 2018-08-15:
Oh. And since we have now one repo per board, we can call the git tags just "v1.3" without the "Zotino/" prefix.

hartytp, 2018-08-15:
> I consider the v1.1 and v1.2 designs also "production". They just have an error that requires rework.

True. But, they way I look at it, if you apply the rework to those designs then they are effectively v1.3 boards. Anyway, I didn't really mean much by that comment other than that I'm happy the bugs seem to be fixed and it works really nicely