sinara-hw/Zotino #20: Power cycling problem

krmac, 2019-05-16:
I've noticed that some boards have problem with quick power cycling. Here is the example:

1. Power supply is on, board is working correctly, consumes about 190 mA.
2. I switch power supply off.
3. I switch power supply on after e.g. 3 seconds. Now board consumes 90 mA, there is +2V on +Vout and -2V on -Vout instead of +13V and -13V.

Here is the measurement on -Vout (CC6-1212DF-E):

[image]

To get the board to work again it is needed to switch power supply off, wait for up to 10 seconds until capacitors are discharged and after switching power supply on again board will work correctly.

hartytp, 2019-05-16:
Which hw revs?

krmac, 2019-05-16:
v1.3

gkasprow, 2019-05-16:
It looks like the DC/DC converter module does not like the power transients and needs the output voltage to fall to zero before input can be re-applied.
We can increase the load or add transistor circuit that discharges output capacitors when input power was removed.

hartytp, 2019-05-16:
> or add transistor circuit that discharges output capacitors when input power was removed.

If we decide to do that, we should consider what other designs need this feature as well.

hartytp, 2019-05-16:
Might be worth power cycling one of these supplies a few times with the loads (other than the capacitors) disconnected to check that it really is the TDK supply failing to start with an output voltage already there.

gkasprow, 2019-05-16:
Let's add 1k resistor to the problematic rail and see if that helps.

hartytp, 2019-09-07:
I've done quite a few design reviews for Sinara recently and a decent fraction of the SMPSs/LDOs we use will explicitly not start correctly into pre-biassed loads. Then there are also the issues around getting power sequencing right with pre-biassed loads.

FWIW quite a lot of commercial equipment will also misbehave if the power is cut very briefly IME. I think the amount of extra complexity, design and testing work that would be required to make Sinara hw always start reliably after brief power cuts is not worth the benefit; it's a surprisingly messy thing to make work.

Note that in this specific case, it's not even 100% clear if the issue is really with the main +-12V SMPS not starting correctly, or with some component that follows it misbehaving and then causing the regulator to current limit, so more debugging would be needed to identify the appropriate solution.

In general I'm also against adding bleed resistors, since they're typically either to small to do anything or large enough to waste a significant amount of power. e.g. Zotino already has quite a bit of quiescent current draw from these rails, so I don't think a resistor will do much (unless your model is that the current draw cuts out when the voltage drops below some ULV threshold, but this threshold is still high enough to bias the SMPS enough to prevent proper startup, but that theory would need testing).

hartytp, 2019-09-07:
IMO a better solution here is documentation. e.g. having a high-level "treating your sinara cards with love" that includes basic things like not rapidly (<1s) power cycling them, powering then on when lying on metal (an apparent student favourite), etc

closed 2019-09-07 by hartytp

hartytp, 2019-09-07:
To put it another way, no power cycling should ever cause damage to a Sinara system. That would be a very serious bug IMO.

However, guaranteeing startup even with odd power cycling doesn't have good effort to reward ratio IMO. Even if we sort out all the LDOs there can be issues around ICs like DDSs, PLLs, etc locking up if not powered up correctly. That's really hard to fix if the boards start up with some capacitors charged.

hartytp, 2019-09-07:
Not to mention the fact that all rails should already have bleeds provided by the indicator leds (but if any of those are missing, that would warrant an issue)

gkasprow, 2019-09-07:
Recently my colleague broke programmable load by too fast power cycling. The fuse went off and we had to send it for reparation.