sinara-hw/Zotino #24: Missing: IDC Connector Specs/Pitch

drewrisinger, 2019-08-07:
The IDC connector pitch & number of positions is not specified in either this project's Wiki/README's, nor [BNC-IDC](https://github.com/sinara-hw/BNC_IDC).

Visually it looks like it's a 2x13 @ 0.1" pitch polarized IDC connector, but I would prefer to have some confirmation before buying appropriate connectors.

**Request**: update the Wiki of both Zotino & BNC_IDC with appropriate connector information.

sbourdeauducq, 2019-08-08:
Those are very common connectors in electronics. If you want an easy way to obtain a Sinara system, you can purchase a pre-assembled crate including bitstreams, device database, etc.
Otherwise, writing this kind of basic documentation is a good first contribution.

drewrisinger, 2019-08-08:
@sbourdeauducq they are a common connector TYPE, but it's also simultaneously vague. Like specifying "FPGA" or "IC". What package do those refer to? `DIP != QFN != SOIC != TO`. https://en.wikipedia.org/wiki/Insulation-displacement_connector.

Based on @sbourdeauducq's very helpful response, and hunting down the BOM hidden under [releases](https://github.com/sinara-hw/Zotino/releases), I see that they are .1" x .1" 2x13 polarized IDC connectors (Socket = 3M_N2526-6002RB).

Wiki editing is limited to members of the `sinara-hw` Github group, so it's not a contribution that I can actually make.

gkasprow, 2019-08-09:
@drewrisinger if you want, I can give you access rights.

drewrisinger, 2019-09-03:
@gkasprow Sure, I can edit a few things. It looks like the UMD section of https://github.com/sinara-hw/meta/wiki/Team also needs to be updated because postdocs and staff have moved on (https://github.com/sinara-hw/Fastino/issues/2#issuecomment-526807081)

hartytp, 2019-09-03:
>  It looks like the UMD section of https://github.com/sinara-hw/meta/wiki/Team also needs to be updated because postdocs and staff have moved on (sinara-hw/Fastino#2 (comment))

I'd be more inclined to just scrap that section. Sinara usage has grown so fast that it's now very non-representative of the actual user base. It's also the kind of thing that tends to go out of date faster than it's updated so unlikely to be much help.

drewrisinger, 2019-09-03:
@hartytp @jbqubit I have no preference on the wiki/Team section, it's interesting and useful for people on-boarding to get an idea of the team members, but it does fall out of date.

closed 2021-11-05 by gkasprow